HefnySco/OBAL #14: sbus support and battery sensor

manotroll, 2022-04-30:
is it possible to add support for sbus or lora ?
sbus as it supports more channels and blonde for range via serial interface
this board could have a current/voltage meter on the board itself and also an output for each "S" in individual so you don't need the external Frsky mlvss sensor

HefnySco, 2022-05-01:
for SBUS: when you enable PPM -which is available by enabling some DEFINE in code there is a function called _process_sbus_pulse that should be able to process SBUS signals.

for current/voltage meter the ADS1115 has 4 inputs. driver can be updated to read from all inputs so we can read a 4S cell or a 3S cell + current. But this required code modification.

I will check the possibility of this, it is a very useful idea as it will save an unnecessary device.

manotroll, 2022-05-05:
ok to use this card and better a pi3 or pi 4 ?
or is pi zero possible to use?

HefnySco, 2022-05-06:
Actually PI-Zero works great. and if you are going to use GY-91 version then use Pi-Zero 2.

HefnySco, 2022-05-21:
please refer to this [discussion](https://github.com/HefnySco/OBAL/discussions/11) there is a simple hack allows you to connect LIPO cells to ADS1115 without any programming.

HefnySco, 2022-07-21:
Regarding SBUS please refer to this post on OBAL [https://discuss.ardupilot.org/t/obal-a-simple-linux-based-raspberry-pi-ardupilot-fcb/75479/25?u=mhefny](https://discuss.ardupilot.org/t/obal-a-simple-linux-based-raspberry-pi-ardupilot-fcb/75479/25?u=mhefny)

HefnySco, 2023-03-17:
BTW you can use breakout like this one:

https://www.aliexpress.com/item/1005001892137682.html?spm=a2g0o.order_list.order_list_main.11.29361802B0MBzR

[image]

closed 2023-03-17 by HefnySco