HefnySco/OBAL #15: fritzing file of the OBAL board

pghpete, 2022-05-05:
I'm in the process of soldering up my board, attaching components and determining additional wiring. I was wondering if you had a fritzing file for this board? It would be extremely useful.  If you don't or don't have the time, I may try to find the time to learn how to convert this or generate one.

HefnySco, 2022-05-06:
I use [easyeda.com](easyeda.com)
my boards are shared at [https://oshwlab.com/mohammad.hefny?tab=project&page=1](https://oshwlab.com/mohammad.hefny?tab=project&page=1)
It is an easy free app BTW.

pghpete, 2022-05-09:
Thank you.  This is more for wiring various sensors to the OBAL board not the actual PCB design. If I do get the chance to create it, i'll share it here in case you or anyone else is interested in it.

HefnySco, 2022-05-09:
Thank you. Please share your design if possible.

> This is more for wiring various sensors to the OBAL board not the actual PCB design.
Are you going to add extra sensors ?

pghpete, 2022-05-09:
I would like to see if I can't use the MPU9250 via i2c since I cannot seam to find code to properly use it on the rpi via spi.  I have also added a UART expansion board HAT and a witty pi 3 (power, safe shutdown, and RTC) board to it.  I believe i'll only need to move two pins via code, 
 OBAL LED_3 move to RC8 (BCM 16)
OBAL SRVO_EN move to RC7 (BCM 20)

I'll lose 2 RC channels but they shouldn't be needed for a boat with only 2 thrusters and differential steering. I'll keep you posted on the project.

HefnySco, 2022-05-09:
You can connect MPU9250 via I2C but it will not work properly because i2c is shared with other sensors. Using SPI you can use MPU9250.

pghpete, 2022-05-09:
It doesn't conflict with any of the existing i2c addresses already in use... that I can see. (unless i'm missing one?)  Does the ardurover code have the spi code in it for the mpu or is it a separate python module or other code?

[uav-boat-pinout](https://drive.protonmail.com/urls/EKA4S1ZCR8#wMEV3zAyztGj)

HefnySco, 2022-05-09:
> It doesn't conflict with any of the existing i2c addresses already in use... that I can see. (unless i'm missing one?) Does the ardurover code have the spi code in it for the mpu or is it a separate [...]

It does not conflict. It just sometimes gives IMU temp reset much more often than SPI.

pghpete, 2022-05-09:
ah. gotcha. ty.

closed 2022-06-13 by HefnySco