HefnySco/OBAL #17: error

leelaser, 2022-07-21:
raspberry pi4 8G ubuntu20.04 server
<pre>SOC: Broadcom BCM2711    use intern: 3 
Failed to get GPIO memory map.
Segmentation fault
</pre>

leelaser, 2022-07-21:
> raspberry pi4 8G ubuntu20.04 server
> 
> SOC: Broadcom BCM2711    use intern: 3 
> Failed to get GPIO memory map.
> Segmentation fault

/dev/mem can opened, but

mmap: Operation not permitted

HefnySco, 2022-07-21:
you are running arducopter using sudo ? it should run as sudo.

leelaser, 2022-07-21:
yes, sudo ./arducopter

leelaser, 2022-07-21:
yes , sudo ./arducopter .then
SOC: Broadcom BCM2711 use intern: 3
Failed to get GPIO memory map.
Segmentation fault

/dev/mem can opened, but

mmap: Operation not permitted

leelaser, 2022-07-21:
yes, using 
sudo ./arducopter  

SOC: Broadcom BCM2711 use intern: 3
Failed to get GPIO memory map.
Segmentation fault

/dev/mem can opened, but

mmap: Operation not permitted

HefnySco, 2022-07-21:
Did you try to run it on Raspberry OS instead of Ubuntu ?
I am trying to understand if it is an OS related issue or HW related issue.

leelaser, 2022-07-21:
pi@raspberrypi:~ $ sudo ./arducopter
sudo: unable to execute ./arducopter: No such file or directory

raspberry pi4,8G .RaspberryPi OS Lite 64-bit

leelaser, 2022-07-21:
Does the compiler need to be replaced with aarch64-linux-gnu?

HefnySco, 2022-07-21:
[https://github.com/HefnySco/OBAL/blob/main/SourceCode.md](https://github.com/HefnySco/OBAL/blob/main/SourceCode.md) should work with RPI-Zero & 4 on Raspberry OS.
I have never tested it on Ubuntu.

HefnySco, 2022-07-21:
> Does the compiler need to be replaced with aarch64-linux-gnu?

try to compile ardupilot on the RPI itself. It will take time but at least you will be sure that there is no compiler issues.

leelaser, 2022-07-22:
hi HefnySco, I am using raspberry Pi4, 8G memory, 64-bit raspberryPi Lites OS, compile obal, need to use aarch64-linux-gnu to run, but error Segmentation fault
, single-step mmap: Operation not permitted, can you tell me the raspberry PI4 you are using, how much memory is there, and whether the os system is 64-bit? Because your precompiled arducopter doesn't work on my platform.

HefnySco, 2022-07-22:
please contact me on my email mohammad.hefny@gmail.com
I can use AnyDesk app and access your Raspberry and figure out the issue -I hope -.

HefnySco, 2022-07-22:
Hi,
I have installed  aarch64 and compiled ardupilot on RPI-4 (4GB)
using the following command
./waf configure --toolchain=native --board=obal

it gave me segmentation error.
seems HAL/Linux in ardupilot does not support aarch64-linux-gnu

leelaser, 2022-07-22:
Hi，my raspberry support aarch64-linux-gnu.
I solved the problem of mmap: Operation not permitted, because there is no PROT_EXEC in the ubuntu20.04 mmap() flag and port, so exec cannot appear in mmap.

At the same time, in raspberry PI4B 64-bit OS, the problem of mmap: Operation not permitted is also solved.
uint32_t GPIO_RPI::get_address(GPIO_RPI::Address address, GPIO_RPI::PeripheralOffset offset) const
{
     return static_cast<uint32_t>(address) + static_cast<uint32_t>(offset);
}

GPIO_RPI::Address address, use 0xFE000000 instead
GPIO_RPI::PeripheralOffset offset , use 0x200000 instead
Others do not need to be changed

leelaser, 2022-07-22:
Hi,HefnySco:
I also try your method

HefnySco, 2022-07-22:
Thank you for your feedback.

one thing I find it strange. when the app failed in this [link](https://github.com/HefnySco/OBAL/issues/17#issuecomment-1191397026) the output was
SOC: Broadcom BCM2711 use intern: **3**
3 is the value returned from UtilRPI::_check_rpi_version_by_rev() and based on it 
`peripheral_base = Address::BCM2711_PERIPHERAL_BASE;` 
which is 
`BCM2711_PERIPHERAL_BASE = 0xFE000000,`

also 

```
enum class PeripheralOffset : uint32_t {
        GPIO = 0x200000,
    };
```
0x200000 is the only option.

I think there is something I do not understand. if you have this on github please share it with me, also you can submit it as pr to ardupilot.

HefnySco, 2023-02-01:
Finally I have succeeded to run OBAL on Bullseye64 but you need to recomile linux kernel and enable `CONFIG_STRICT_DEVMEM=y`
steps in detailed are [here](https://medium.com/@mohammad.hefny/running-ardupilot-on-raspberry-os-64-bullseye-832f7d28bc22)

closed 2023-02-01 by HefnySco