HefnySco/OBAL #18: Cross Compilation fail

maaz-shahid99, 2022-07-21:
[671/865] Compiling libraries/AP_ADSB/sagetech-sdk/toAlt.c
[672/865] Compiling libraries/AP_ADSB/sagetech-sdk/uint322Buf.c
[673/865] Compiling libraries/AP_ADSB/sagetech-sdk/toTOA.c
[674/865] Compiling libraries/AP_ADSB/sagetech-sdk/toIcao.c
[675/865] Compiling libraries/AP_ADSB/sagetech-sdk/icao2Buf.c
[676/865] Compiling libraries/AP_ADSB/sagetech-sdk/toUint32.c
[677/865] Compiling libraries/AP_ADSB/sagetech-sdk/toVel.c
[678/865] Compiling libraries/AP_ADSB/sagetech-sdk/uint162Buf.c
../../libraries/AP_HAL_Linux/Thread.cpp: In member function ‘bool Linux::Thread::set_stack_size(size_t)’:
../../libraries/AP_HAL_Linux/Thread.cpp:248:44: error: ‘PTHREAD_STACK_MIN’ was not declared in this scope
     _stack_size = MAX(stack_size, (size_t) PTHREAD_STACK_MIN);
                                            ^~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.

Waf: Leaving directory `/home/maaz/Tdisk/obal/ardupilot/build/obal'
Build failed
 -> task in 'objs/AP_HAL_Linux' failed (exit status 1): 
	{task 140199083980160: cxx Thread.cpp -> Thread.cpp.0.o}
 (run with -v to display more information)

maaz-shahid99, 2022-07-21:
> `[671/865] Compiling libraries/AP_ADSB/sagetech-sdk/toAlt.c [672/865] Compiling libraries/AP_ADSB/sagetech-sdk/uint322Buf.c [673/865] Compiling libraries/AP_ADSB/sagetech-sdk/toTOA.c [674/865] [...]

Error encountered when compiling using this command:
./waf configure --toolchain=/opt/cross-pi-gcc/bin/arm-linux-gnueabihf --board obal

HefnySco, 2022-07-22:
The issue in the toolchain.

please download this particular [file](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%2010.2.0/Raspberry%20Pi%201%2C%20Zero/) 

expand it in /opt/cross-pi-gcc-10.2.0-0 

sudo ln -s /opt/cross-pi-gcc-10.2.0-0    /opt/cross-pi-gcc

if you still face problems pls let me know I can help you using remote desktop.

closed 2022-08-27 by HefnySco