HefnySco/OBAL #29: Compiling directly on Rapberry

highfreq, 2023-02-18:
Since people will run different OS on their pc, wouldn't it be easier to just give instructions to compile on the raspberry that everyone will use?

HefnySco, 2023-02-18:
compiling it on the raspberry it sraight forward.... download ardupilot
then make obal.
that is it :)

highfreq, 2023-02-18:
Any chance to put it on the how to?

HefnySco, 2023-02-18:
please try this [procedure](https://ardupilot.org/dev/docs/building-setup-linux.html#building-setup-linux)
then 
`make obal`

highfreq, 2023-02-19:
My point is, since users will have all different PC but will use all a raspberry, why just don't give instructions to compile on the Raspberry that is the common platform everybody will use, regardless of the PC they use and the crosscompile mess.

highfreq, 2023-02-19:
Tried the procedure linked, i got this error when configuring for obal board:

pi@raspberrypi:~/ardupilot $ ./waf configure --board obal
Setting top to                           : /home/pi/ardupilot
Setting out to                           : /home/pi/ardupilot/build
Autoconfiguration                        : enabled
Setting board to                         : obal
Using toolchain                          : arm-linux-gnueabihf
Checking for 'g++' (C++ compiler)        : not found
Checking for 'clang++' (C++ compiler)    : /usr/bin/clang++
Checking for 'gcc' (C compiler)          : /usr/bin/arm-linux-gnueabihf-gcc
Checking for c flags '-MMD'              : yes
Checking for cxx flags '-MMD'            : yes
CXX Compiler                             : clang++ 10.2.1
Checking for need to link with librt     : necessary
Checking for library rt                  : not found
The configuration failed
(complete log in /home/pi/ardupilot/build/config.log)

highfreq, 2023-02-19:
I am trying to compile obal bin on a raspberry Bulleys, maybe that is the problem, i should try on llegacy.

HefnySco, 2023-02-19:
pip install rt

highfreq, 2023-02-19:
Same error after pip install rt

HefnySco, 2023-02-19:
I did this steps and it was compiled successfully

git clone -b master --single-branch
https://github.com/ArduPilot/ardupilot.git --depth 1
cd ardupilot
./Tools/gittools/submodule-sync.sh
sudo apt install python-pip
python -m pip install empy
python -m pip install pexpect
python -m pip install future
make obal

highfreq, 2023-02-19:
I get the same error, maybe it is because i am using bullseye, do you think i should try legacy?

HefnySco, 2023-02-19:
most probably
yes.

Also bullseye will not work with obal unless you rebuild the kernel
***@***.***/running-ardupilot-on-raspberry-os-64-bullseye-832f7d28bc22

highfreq, 2023-02-19:
Yes i read it, i was just using it to build firmware but i'll try with legacy and hopefullyit'll work.

thanks for your help.

highfreq, 2023-03-14:
I can confirm that using legacy Buster everything worked as expected.

closed 2023-03-14 by HefnySco