HefnySco/OBAL #30: Running On Banana Pi M2 Zero

xagawar, 2023-02-19:
Is it possible to use OBAL with Banana Pi M2 Zero?

HefnySco, 2023-02-19:
I have made some modification to ardupilotcode to run BananaPi W2
https://github.com/HefnySco/ardupilot/tree/pr_banana_zero
my only problem is RCInput.
It you can replace it with RC_UDP ot will work.
Now I am working to solve this problem. I will publish the solution once
completed.

kar1150, 2023-06-07:
what OS is recommended on the banana, I on this -  Raspbian GNU/Linux 10 (buster) armv7l, Kernel: 4.19.129-sunxi 
it does not see the baro - Config Error: Baro: unable to initialise driver

I tried a raspberrypi binary on the same OS and it seems to run, but is very very sloppy with gyro response in Missionplanner
thank you in advance!

DroneEngage, 2023-06-07:
I havent completed this solution yet. I depend on a STM32F411 microcontroller that act as slave I2c and connect sensors to it.
but again still facing problems.