HefnySco/OBAL #31: gy-91 patch as a branch or a different board

awright424, 2023-02-19:
is it a possibility of creating a branch or different board for the gy-91 I get this error trying to apply your patch.

> alex@DESKTOP-ITUSAHU:~/ardupilot$ git apply obal_gy91.patch
error: corrupt patch at line 35

or fix the patch. I downloaded the files to the root directory of the ardupilot files and got that error when applying the patch

HefnySco, 2023-02-19:
fixed.... thank you for highlighting this issue.
Actually I prefer to leave it as a batch because OBAL is a part if Ardupilot release, so you should always be able to use latest updates and run this patch to convert OBAL-IMU-9250 to OBAL-GY91.

awright424, 2023-02-19:
works for me now. I am getting a config error. I bought my modules from amazon and you mentioned the led should not be red so it may be that I have a fake

[image]

awright424, 2023-02-19:
one of the imus i ordered was bad.

closed 2023-02-19 by awright424