HefnySco/OBAL #36: Testing waiting for hardware

highfreq, 2023-03-15:
I am waiting for my board and sensor that have been ordered. I installed obal on a raspberry 3. Everything went well but when i start ardupilot after a minute or two the raspberry becomes unresponsive and have to reboot to ssh to it. Can it be because there is still no obal board on it?

HefnySco, 2023-03-15:
I do not know. I have tested OBAL on RPI-Zero,Zero2 & RPI-4 only. 
But I can say that I left OBAL working all the night many times as sort of stress test and it was working fine.
try to run using GDB command on ssh and see how it crashes. 
1- Which obal configuration are you using ?
2- Have you made any changes to the code ?

highfreq, 2023-03-15:
1 - We made no change to the code
2 - I am using 4.3 rover with gy-91 patch (i asked you the correct procedure to apply patch in another issue thread).
3 -We DO NOT have any obal hardware attached to the raspberry yet, just the software running and it crashes.

highfreq, 2023-03-15:
Maybe is the param file, how can i create a base param file for rover?

highfreq, 2023-03-15:
I created a parm file as you adviced on the wiki and it doesn't crash anymore.

closed 2023-03-17 by HefnySco