HefnySco/OBAL #39: Bro config issue

highfreq, 2023-03-24:
I get this error in mission planner when connecting. I am using a GY-91 board:

 Config Error: Baro: unable to initialise driver

highfreq, 2023-03-24:
Will retry, do i have to download dev branch?

highfreq, 2023-03-24:
I downloaded and applied patch, and got no error. I am now compling.

Will let you know how it goes.

highfreq, 2023-03-25:
Could it be the raspberry 3 not working?
I doubt it because the battery and pwm boards are correctly picked up by the software, it is only the GY-91 not working.

highfreq, 2023-05-24:
@HefnySco can you please elaborate a bit more on where to disable baro? I need to use OBAL on a boat so no baro needed and i get the baro init error.
My board is GY-91 type.

highfreq, 2023-05-24:
what file do i have to modify?

HefnySco, 2023-05-24:
> what file do i have to modify?

libraries/AP_HAL/board/linux.h

highfreq, 2023-05-24:
But if i want to use the barometer on the gy-91 what should i do to not have the barometer error?

highfreq, 2023-05-24:
what i mean is, i think there is a barometer on the gy-91 imu, can i use it via spi? if so, how can it be used?

highfreq, 2023-05-25:
if i uncomment the following will the baro on the gy-91 work???

//#define HAL_BARO_PROBE_LIST PROBE_BARO_SPI(BMP280, "bmp280")

HefnySco, 2023-05-31:
I just discovered a modification in Ardupilot that corrupts OBAL
please use my version [https://github.com/HefnySco/ardupilot/tree/pr_fix_RPI_check](https://github.com/HefnySco/ardupilot/tree/pr_fix_RPI_check)  until fix is approved.

HefnySco, 2024-03-28:
approved [https://github.com/ArduPilot/ardupilot/pull/23943](https://github.com/ArduPilot/ardupilot/pull/23943)

closed 2024-03-28 by HefnySco