HefnySco/OBAL #46: Connect Mission Planner to OBAL

akhodeir, 2023-10-25:
Hi,

I would appreciate if you can explain more,  how to connect mission planner to OBAL. 

for normal boards(STM32Fxxx), we [connect](https://ardupilot.org/planner/docs/common-connect-mission-planner-autopilot.html) the board to the computer that has "mission planner" installed on it. 

For OBAL boards, how shall we connect the board to the computer for calibaration,..etc ? (cables, wifi , ... etc ? )

okonon, 2024-03-27:
@akhodeir  i am interested in this as well. Have you gotten the answer to your question ?

HefnySco, 2024-03-28:
Hi,
Sorry for this delay. I missed the question.
check this [link](https://github.com/HefnySco/OBAL/wiki/06---Build-and-Install#running-service) please.
`ExecStart=/home/pi/ardurover -A  udp:**YourTargetIP**:14550:bcast -B /dev/serial0
`
1- you can set your IP to match your target GCS.
2- Another BETTER solution is to make your raspberry run with a static IP and then you connect to it but then you need to change the command to 
`ExecStart=/home/pi/ardurover -A  udpin:0.0.0.0:14550:bcast -B /dev/serial0`

3- An even better solution is to run RPI as an access point with its own WIFI network and you can connect to it from any computer without the need to use a router in between. check this [code](https://github.com/DroneEngage/DroneEngage_ScriptWiki/blob/main/helper_scripts/create_ap.sh)  from my project DroneEngage which can be easily run on the same board BTW.