HefnySco/OBAL #48: New option for Raspberry Pi

happytm, 2023-11-10:
Hi,
I have been following your work since last ten years. I wanted to see if OBAL can work with linked linux board as a flight controller?

https://www.youtube.com/watch?v=Mghr4vG-SR0

Thanks.

HefnySco, 2023-11-13:
Hi,
Thank you for following me.
Actually I am looking into integrating Linux based boards with multiple small STM32 microprocessors to make use of computational power of large boards and IO of STM32.
There are some topics here this is one of them [https://medium.com/@mohammad.hefny/ardupilot-linux-based-board-stm32-as-a-breakout-module-5b57989863ce](https://medium.com/@mohammad.hefny/ardupilot-linux-based-board-stm32-as-a-breakout-module-5b57989863ce)

okonon, 2024-03-06:
@HefnySco could you please point me to right direction where i can find final board design based on STM32 that you write in your blog ? thanks

HefnySco, 2024-03-28:
I havent made a dedicated PCB yet for it. You need to connect it next to OBAL.

okonon, 2024-04-01:
@HefnySco understood thank you