HefnySco/OBAL #49: Could not find the program ['/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-ar']

sanojan2008, 2024-02-19:
Could not find the program ['/opt/cross-pi-gcc/bin/arm-linux-gnueabihf-ar']

After Running these Commands
(Compilled directly on the RPI)

git clone -b master --single-branch
https://github.com/ArduPilot/ardupilot.git --depth 1
cd ardupilot
./Tools/gittools/submodule-sync.sh
sudo apt install python-pip
python -m pip install empy
python -m pip install pexpect
python -m pip install future
make obal

./waf configure --toolchain=/opt/cross-pi-gcc/bin/arm-linux-gnueabihf --board obal

[image]

HefnySco, 2024-02-19:
you need to install it from [https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%2010.2.0/Raspberry%20Pi%201%2C%20Zero/](https://sourceforge.net/projects/raspberry-pi-cross-compilers/files/Raspberry%20Pi%20GCC%20Cross-Compiler%20Toolchains/Buster/GCC%2010.2.0/Raspberry%20Pi%201%2C%20Zero/)

HefnySco, 2024-02-19:
please check https://github.com/HefnySco/OBAL/wiki/06---Build-and-Install

sanojan2008, 2024-02-19:
Thanks for the fast reply 
Amazing Work
Will try this out now

sanojan2008, 2024-02-19:
May I ask how to install the tar file on SSH RPI4

HefnySco, 2024-02-19:
This is beyond the scope of this thread. Please google for these linux related commands.

closed 2024-02-19 by HefnySco

sanojan2008, 2024-02-19:
Alright thanks

sanojan2008, 2024-02-19:
Ive downloaded the Compiler
It still says it cant be found
[image]
[image]

Please Help me Ive literally spent 3 days and night trying to fix this

HefnySco, 2024-02-20:
Please /opt is a root folder not inside ardupilot
please you can consult poe.com for this linux issue.