HefnySco/OBAL #50: Failed phys2virt prev_virt=0xf7f2f000 phys_page=0xc5e20000 Operation not permitted

sanojan2008, 2024-02-19:
[image]

Any Ideas why Im getting this error
I feel like I'm using a wrong version of RPI OS
but when I install Buster it says the the python version is too old

HefnySco, 2024-02-20:
You need to recompile Linux to allow CONFIG_STRICT_DEVMEM = ENABLED.

HefnySco, 2024-02-20:
Compile Linux 64-bit
https://www.raspberrypi.org/documentation/computers/linux_kernel.html#configuring-the-kernel

Detailed Description
https://lemariva.com/blog/2019/09/raspberry-pi-4b-preempt-rt-kernel-419y-performance-test

sanojan2008, 2024-02-21:
Ive finished building there kernel using Rasbperry pi website the detailed one isn't working 
I'm compiling the files directly on the RPI4 as I don't have linux and don't want it

HefnySco, 2024-02-22:
https://docs.google.com/document/d/1uomo_2-cMGGrQfBMLpcBvQ17OGtd19-J6zNGcjWSems/edit?usp=sharing

This is my own How-To document... it is not well written but you should be able to find your way through it.

HefnySco, 2024-02-25:
> > https://docs.google.com/document/d/1uomo_2-cMGGrQfBMLpcBvQ17OGtd19-J6zNGcjWSems/edit?usp=sharing
> > This is my own How-To document... it is not well written but you should be able to find your way through it.
> 
> Hello thank yoy for sharing this I've requested access to view this file could you please accept it thank you

done

okonon, 2024-03-27:
@HefnySco i requested access as well. Would you be able to grant it ? Thanks a lot. Also is it possible to get latest stm32 
board info?

closed 2024-03-28 by HefnySco