HefnySco/OBAL #54: OBAL startup problem. (SIGSEGV, Segmentation fault.)

AlexanderBVKtex, 2024-06-17:
## Bug report

**Issue details**
After running my compiled executable for OBAL, the following error appears: SIGSEGV, Segmentation fault. (Everything is done according to OBAL's [instructions](https://github.com/HefnySco/OBAL/wiki/06---Build-and-Install).)
When debugged, it gives the following output:
```
RPI 4 

Program received signal SIGSEGV, Segmentation fault.
AP::rtc () at ../../libraries/AP_RTC/AP_RTC.cpp:309
309	{
```

**Version**
Ardupilot 4.5.4; No problems on Ardupilot 4.3.3

**System**
Ubuntu 22.04 arm64
Compiled with clang++ 14.0.0

HefnySco, 2024-06-19:
Thank you for highlighting this issue.

Please try to run with the following update:
**libraries/AP_RTC/AP_RTC_config.h**
```
#ifndef AP_RTC_ENABLED
#define AP_RTC_ENABLED 0
#endif
```

HefnySco, 2024-06-19:
Please confirm that CONFIG_STRICT_DEVMEM is enabled.
as mentioned here [https://github.com/HefnySco/OBAL/wiki/06---Build-and-Install](https://github.com/HefnySco/OBAL/wiki/06---Build-and-Install).

I made a fast test on RPI-Zero W2 with 2021-05-07-raspios-buster-armhf-lite.zip and every thing seems to be working fine.

AlexanderBVKtex, 2024-06-19:
I recompiled and installed the kernel with a modified CONFIG_STRICT_DEVMEM parameter for ubuntu. Access to /dev/mem is available. If necessary, [this is my kernel](https://cert.bvksite.com/kern.zip). (linux-raspi-5.15.0-1017-arm64)

AlexanderBVKtex, 2024-06-19:
> Thank you for highlighting this issue.
> 
> Please try to run with the following update: **libraries/AP_RTC/AP_RTC_config.h**
> 
> ```
> #ifndef AP_RTC_ENABLED
> #define AP_RTC_ENABLED 0
> #endif
> ```

From this modification I realised that the problem is that I am compiling with the `--enable-dds` parameter, which gives an error **SIGSEGV, Segmentation fault** at startup. Without this parameter everything works fine. But the point is that this parameter is necessary for ros2 to work.

HefnySco, 2024-06-19:
I am not aware of the --enable-dds function but seems to be it is related to SITL.

AlexanderBVKtex, 2024-06-19:
This function is needed for [ROS2](https://www.ros.org/blog/why-ros/) operation (thanks to ros it is possible to create a drone that works completely autonomously). This parameter appeared relatively recently and is used in binary files building. (adding this parameter during compilation is described [here](https://ardupilot.org/dev/docs/ros2-over-ethernet.html)).

HefnySco, 2024-06-19:
The code crashes here.

```
int clock_gettime(clockid_t clockid, struct timespec *ts)
{
    //! @todo the value of clockid is ignored here.
    //! A fallback mechanism is employed against the caller's choice of clock.
    uint64_t utc_usec;
    if (!AP::rtc().get_utc_usec(utc_usec)) {
        utc_usec = AP_HAL::micros64();
    }
    ts->tv_sec = utc_usec / 1000000ULL;
    ts->tv_nsec = (utc_usec % 1000000ULL) * 1000UL;
    return 0;
}

```

Program received signal SIGSEGV, Segmentation fault.
0x0023844c in Linux::Scheduler::from (
    scheduler=<error reading variable: Cannot access memory at address 0x7e7ffffc>) at ../../libraries/AP_HAL_Linux/Scheduler.h:24
24	../../libraries/AP_HAL_Linux/Scheduler.h: No such file or directory.
(gdb) stack
Undefined command: "stack".  Try "help".
(gdb) backtrace
#0  0x0023844c in Linux::Scheduler::from (
    scheduler=<error reading variable: Cannot access memory at address 0x7e7ffffc>) at ../../libraries/AP_HAL_Linux/Scheduler.h:24
#1  0x0024a320 in AP_HAL::micros64 ()
    at ../../libraries/AP_HAL_Linux/system.cpp:62
#2  0x002899b0 in clock_gettime (clockid=1, ts=0x7e800048)
    at ../../libraries/AP_DDS/AP_DDS_Client.cpp:1169
#3  0x0024a354 in AP_HAL::micros64 ()
    at ../../libraries/AP_HAL_Linux/system.cpp:69
#4  0x002899b0 in clock_gettime (clockid=1, ts=0x7e800088)
    at ../../libraries/AP_DDS/AP_DDS_Client.cpp:1169
#5  0x0024a354 in AP_HAL::micros64 ()
    at ../../libraries/AP_HAL_Linux/system.cpp:69
#6  0x002899b0 in clock_gettime (clockid=1, ts=0x7e8000c8)
    at ../../libraries/AP_DDS/AP_DDS_Client.cpp:1169
#7  0x0024a354 in AP_HAL::micros64 ()
    at ../../libraries/AP_HAL_Linux/system.cpp:69
#8  0x002899b0 in clock_gettime (clockid=1, ts=0x7e800108)
    at ../../libraries/AP_DDS/AP_DDS_Client.cpp:1169
#9  0x0024a354 in AP_HAL::micros64 ()
    at ../../libraries/AP_HAL_Linux/system.cpp:69
#10 0x002899b0 in clock_gettime (clockid=1, ts=0x7e800148)
    at ../../libraries/AP_DDS/AP_DDS_Client.cpp:1169
--Type <RET> for more, q to quit, c to continue without paging--

const Linux::Scheduler* scheduler = Linux::Scheduler::from(hal.scheduler);

AlexanderBVKtex, 2024-06-23:
I can suggest a **crutch** for the file: `libraries/AP_HAL_Linux/system.cpp`
```
#include <stdarg.h>
#include <stdio.h>
#include <sys/time.h>
#include <unistd.h>
#include <sys/time.h>

#include <AP_Common/AP_Common.h>
#include <AP_HAL/AP_HAL.h>
#include <AP_HAL/system.h>
#include <AP_HAL_Linux/Scheduler.h>
#include <AP_Math/div1000.h>

extern const AP_HAL::HAL& hal;
struct timeval start_timer, end_timer;
auto temp11=gettimeofday(&start_timer, NULL);

namespace AP_HAL {

static struct {
    uint64_t start_time_ns;
} state;

static uint64_t ts_to_nsec(struct timespec &ts)
{
    return ts.tv_sec*1000000000ULL + ts.tv_nsec;
}

void init()
{
    struct timespec ts {};
    clock_gettime(CLOCK_MONOTONIC, &ts);
    state.start_time_ns = ts_to_nsec(ts);
}

void WEAK panic(const char *errormsg, ...)
{
    va_list ap;

    va_start(ap, errormsg);
    vdprintf(1, errormsg, ap);
    va_end(ap);
    UNUSED_RESULT(write(1, "\n", 1));

    if (hal.rcin != nullptr) {
        hal.rcin->teardown();
    }
    if (hal.scheduler != nullptr) {
        hal.scheduler->delay_microseconds(10000);
    }
    exit(1);
}

uint32_t micros()
{
    return micros64() & 0xFFFFFFFF;
}

uint32_t millis()
{
    return millis64() & 0xFFFFFFFF;
}

uint64_t micros64()
{

   /* const Linux::Scheduler* scheduler = Linux::Scheduler::from(hal.scheduler);
    uint64_t stopped_usec = scheduler->stopped_clock_usec();
    if (stopped_usec) {
        return stopped_usec;
    }

    struct timespec ts;
    clock_gettime(CLOCK_MONOTONIC, &ts);
    return uint64_div1000(ts_to_nsec(ts) - state.start_time_ns);*/

    gettimeofday(&end_timer, NULL);
    long seconds45 = (end_timer.tv_sec - start_timer.tv_sec);
    long micros45 = ((seconds45 * 1000000) + end_timer.tv_usec) - (start_timer.tv_usec);
    return micros45;
}

uint64_t millis64()
{
    return uint64_div1000(micros64());
}

} // namespace AP_HAL

```